antoineco/broadcom-wl #2: Loading of module "wl" freezes my system

arun-gurung, 2017-05-24:
I can compiled and install the module successfully, but it freezes my system. So far, after several reboots, I have come so long as to select network before it again freezes.

antoineco, 2017-05-24:
@arun-gurung what OS and kernel are you using? Did you blacklist the modules mentioned in the README? Can you provide a kernel crash dump?

arun-gurung, 2017-05-24:
I`m on CentOS 7.3 with kernel 4.11.2-1.el7.elrepo.x86_64. I blacklisted the mentioned modules. I deleted "wl.ko" from /lib/modules and now my system is up again. I don`t know how to provide a kernel crash dump.

antoineco, 2017-05-24:
In some cases `journalctl` may display kernel related errors.

Regarding the crash dump you can take a look at https://access.redhat.com/documentation/en-US/Red_Hat_Enterprise_Linux/7/html/Kernel_Crash_Dump_Guide/chap-introduction-to-kdump.html

Unfortunately without logs I can't tell you much about what may be going on. If you built from the current `patch-linux4.7` branch (with commit 358ee38a) and are running on an x86_64 architecture you should be good to go.

arun-gurung, 2017-05-24:
I downloaded the .zip file and compiled it from there. I will try to install the driver again and provide a kernel crash dump later.

arun-gurung, 2017-05-24:
Here`s the output of journalctl: [journalctl.txt](https://github.com/antoineco/broadcom-wl/files/1026106/journalctl.txt)

antoineco, 2017-05-24:
I see no crash in this dump, only `device (wlp0s20u2u4): Activation: successful, device activated` so your WiFi is up, right?

arun-gurung, 2017-05-24:
Yes, the system detects the Wifi card, but when I try to connect to a network, the system freezes.

antoineco, 2017-05-24:
But you **are** connected:
```
mai 24 17:21:47 macbook-air NetworkManager[953]: <info>  [1495639307.9612] dhcp4 (wlp0s20u2u4): activation: beginning transaction (timeout in 45 seconds)
mai 24 17:21:48 macbook-air NetworkManager[953]: <info>  [1495639308.1158] dhcp4 (wlp0s20u2u4): dhclient started with pid 2416
mai 24 17:21:49 macbook-air dhclient[2416]: DHCPREQUEST on wlp0s20u2u4 to 255.255.255.255 port 67 (xid=0x25457847)
mai 24 17:21:49 macbook-air dhclient[2416]: DHCPACK from 192.168.1.254 (xid=0x25457847)
mai 24 17:21:49 macbook-air NetworkManager[953]: <info>  [1495639309.5265] dhcp4 (wlp0s20u2u4):   address 192.168.1.13
mai 24 17:21:49 macbook-air NetworkManager[953]: <info>  [1495639309.5267] dhcp4 (wlp0s20u2u4):   plen 24 (255.255.255.0)
mai 24 17:21:49 macbook-air NetworkManager[953]: <info>  [1495639309.5267] dhcp4 (wlp0s20u2u4):   gateway 192.168.1.254
mai 24 17:21:49 macbook-air NetworkManager[953]: <info>  [1495639309.5267] dhcp4 (wlp0s20u2u4):   server identifier 192.168.1.254
mai 24 17:21:49 macbook-air NetworkManager[953]: <info>  [1495639309.5267] dhcp4 (wlp0s20u2u4):   lease time 86400
mai 24 17:21:49 macbook-air NetworkManager[953]: <info>  [1495639309.5267] dhcp4 (wlp0s20u2u4):   nameserver '192.168.1.254'
mai 24 17:21:49 macbook-air NetworkManager[953]: <info>  [1495639309.5267] dhcp4 (wlp0s20u2u4):   domain name 'zyxel.localhost'
mai 24 17:21:49 macbook-air NetworkManager[953]: <info>  [1495639309.5267] dhcp4 (wlp0s20u2u4): state changed unknown -> bound
mai 24 17:21:49 macbook-air NetworkManager[953]: <info>  [1495639309.5281] device (wlp0s20u2u4): state change: ip-config -> ip-check (reason 'none') [70 80 0]
mai 24 17:21:49 macbook-air avahi-daemon[864]: Joining mDNS multicast group on interface wlp0s20u2u4.IPv4 with address 192.168.1.13.
mai 24 17:21:49 macbook-air avahi-daemon[864]: New relevant interface wlp0s20u2u4.IPv4 for mDNS.
mai 24 17:21:49 macbook-air avahi-daemon[864]: Registering new address record for 192.168.1.13 on wlp0s20u2u4.IPv4.
mai 24 17:21:49 macbook-air dhclient[2416]: bound to 192.168.1.13 -- renewal in 39978 seconds.
mai 24 17:21:49 macbook-air NetworkManager[953]: <info>  [1495639309.5313] device (wlp0s20u2u4): state change: ip-check -> secondaries (reason 'none') [80 90 0]
mai 24 17:21:49 macbook-air NetworkManager[953]: <info>  [1495639309.5316] device (wlp0s20u2u4): state change: secondaries -> activated (reason 'none') [90 100 0]
```

The problem must occur later, once again without any crash log I am clueless.

arun-gurung, 2017-05-24:
I reinstalled CentOS as my primary system on my Macbook Air and installed kernel 4.11.2 and reinstalled your driver and it all seems to work properly now.

[root@macbook-air ~]# lspci | grep -i broadcom
02:00.0 Multimedia controller: Broadcom Limited 720p FaceTime HD Camera
03:00.0 Network controller: Broadcom Limited BCM4360 802.11ac Wireless Network Adapter (rev 03)
[root@macbook-air ~]#

antoineco, 2017-05-24:
Great, good job :) Hopefully I can make Linux work on my MacBook Air some day too.

arun-gurung, 2017-05-24:
My Macbook Air is from June 2015. Linux should work nicely on it, I have almost all my hardware working now.

arun-gurung, 2017-05-25:
The module has started freezing my system again, so I deleted wl.ko and it works again. I will try to provide a kernel crash dump when I got figured out how to do it.

antoineco, 2017-06-08:
@arun-gurung I finally got a chance to test the Linux kernel 4.11 on my Linux machine for a prolonged period of time and it's been rock stable so far after every reboot.

How does it look on your side?

arun-gurung, 2017-06-09:
I have reinstalled and my system works properly now running kernel 4.11.2. I haven't tested your driver again yet, but I will test it in a while. I hope everything works this time, so I don't have to use my dongle anymore.

arun-gurung, 2017-06-20:
I'm sad to say that I tried the module again and it still hangs my system. I cannot simply use it. I don't know how to provide a kernel crash dump yet. But thanks for the effort.

antoineco, 2017-06-20:
Too bad :(
Maybe give the [kmod-wl](http://download1.rpmfusion.org/nonfree/fedora/development/rawhide/Everything/x86_64/os/repoview/kmod-wl.html) package a try (install the [rpmfusion](http://download1.rpmfusion.org/nonfree/fedora/development/rawhide/Everything/x86_64/os/repoview/rpmfusion-nonfree-release.html) repo first). It uses the same codebase but you never know.

arun-gurung, 2017-06-22:
Thank you, I will try the kmod-wl package soon.

arun-gurung, 2017-07-10:
The weird thing is that the driver works with Linux 3.10.

agrecascino, 2017-08-17:
Can confirms this happens on gentoo livecd as of current date when building wl and performing any network activity. I'm on a macbook air mid 2013.

arun-gurung, 2017-12-11:
Bad news: I tried with kernel 4.14 and it still freezes. Thanks for making the patch to compile the Makefile, but I believe the problem is with the closed source driver from Broadcom, because it works fine with kernel 3.10. I will send a message to Broadcom and see what they say.

antoineco, 2017-12-11:
Works for me on Linux 4.13 x86_64.
Linux 4.14 requires another patch. I'll commit it soon.

arun-gurung, 2017-12-11:
I will try your next patch. 

Do you have a theory why it works on kernel 3.10, but not with 4.x?

antoineco, 2017-12-20:
@arun-gurung I can confirm the module works without extra patching on Linux 4.14.3. Your issue seems to be specific to the Macbook hardware but without a crash dump I can unfortunately not emit any theory.

arun-gurung, 2018-02-13:
I too have been busy, so I have no crashdump. And sending a message to Broadcom is pointless, because my WLAN chip doesn't freeze the system with kernel 3.10. Ergo, the software bug is in the 4.x wl.ko module. :-(

antoineco, 2018-02-13:
I couldn't tell without any logs unfortunately. I'm using this module as we speak, every day, on an ever up-to-date kernel.

Are you absolutely sure it is WiFi related? Did you blacklist the modules mentioned in the README?

arun-gurung, 2018-02-15:
Yes, I'm 97% sure it's the wl.ko module. If you are able to use it with Linux 4.x, maybe I just have a hardware error but not likely because the WLAN chip works with Linux 3.10 and on MacOS.

arun-gurung, 2018-02-15:
OK, it's alright. I`ll see later about a crashdump, thanks. :-)

arun-gurung, 2018-03-27:
Still no kernel crash dump from me, but I updated to kernel 4.15 and now your patched module has been stable for several hours, so I am now very content and can finally ditch my USB wi-fi dongle. Congratulations on a job well done to the developers. :-D

antoineco, 2018-03-27:
That's encouraging! Hope you're not celebrating to fast :D

arun-gurung, 2018-04-13:
Well, now the module started freezing again after updating to kernel 4.16.x.

arun-gurung, 2018-04-14:
I am sorry, having a hard time creating a kernel crash dump. I have configured it and it looks like it's set up, but I get no "vmcore" only a text file, like this and this is the messages I got in the terminal right before it hangs.
[vmcore-dmesg.txt](https://github.com/antoineco/broadcom-wl/files/1910115/vmcore-dmesg.txt)
[image]

antoineco, 2018-04-14:
Looks like the `rtl8812au` module for your Realtek Wifi dongle crashed before `wl`. Could you try blacklisting it for the time being? (see README)

arun-gurung, 2018-04-15:
It didnt` help with blacklisting the rtl8811au.

arun-gurung, 2018-04-17:
Ah, new report: It froze again. Now I have installed the kmod-wl package from RPMfusion and it´s been stable for half an hour. Ah, programming is over my head so I have no idea what the problem could be.

arun-gurung, 2018-12-14:
My solution is this, at least on CentOS 7.6: 

1: Download and compile the driver from Broadcom's official support page. Copy the "wl.ko" to "/lib/modules/*kernel-version*/kernel/drivers/net/wireless".

2: Install the following two packages with "rpm --nodeps":

akmod-wl-6.30.223.271-14.fc27.x86_64.rpm
kmod-wl-6.30.223.271-16.fc27.x86_64.rpm

This works for me and it's been stable for a long time.

ktb83, 2019-11-14:
I figured I'd offer my experience. I had been experiencing freezes on Alpine Linux using this driver with kernel 4.19.80 and BCM4331. I didn't investigate it very far, but I'd usually see kernel panic and soft lockup messages before going down. I knew I had been using this driver version from Debian/Ubuntu for a while without this sort of problem on the same hardware. So, I went through the source available here in the patch-linux4.7 branch comparing it to all the Debian patches being used. If I remember correctly, I added in 6 commits from 6 Debian patches missing in this source. Since then, the driver has been working fine for me.

antoineco, 2019-11-14:
@ktb83 thanks, that's some useful feedback 👍 
I'll check the differences with the Debian package. Did you push your changes somewhere by any chance?

ktb83, 2019-11-14:
@antoineco I have not published the code, it's still just on my local machine here. I'll do a quick check on the commits I added.

ktb83, 2019-11-14:
These are the Debian patches I added in:
03-rename-to-wlan0.patch
04-user_ioctl.patch
05-remove-time-and-date-macros.patch
06-broadcom-sta-6.30.223.248-linux-3.18-null-pointer-crash.patch
07-rdtscl.patch
08-linux47.patch

ktb83, 2019-11-15:
@antoineco Here are the commits -- https://github.com/ktb83/broadcom-wl/commits/patch-linux5.1

arun-gurung, 2019-12-20:
I have emigrated from CentOS to Manjaro. The module still freezes my system. It sounds good with these patches, but how can I download them and apply them?

antoineco, 2019-12-20:
@arun-gurung could you please try building the module from the `extra-patches` branch?
```
$ git fetch
$ git checkout extra-patches
```

arun-gurung, 2019-12-21:
Thanks, but programming is over my head and all I can do is use "git clone". I need somebody to hold my hand, please.

$ git fetch 
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).

git checkout extra-patches
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).

antoineco, 2019-12-21:
You have to execute this _inside_ the repository, _after_ performing a `git clone` (which you should already have done since you already built the driver).

arun-gurung, 2019-12-21:
Oh, yes.

My system used to freeze only seconds after loading the driver. Now it's been stable for 15 minutes. I hope this is a good present for the season and I can ditch my USB dongle. Merry Christmas!

antoineco, 2019-12-21:
Good to hear! Merry Christmas to you too 🙌

PR #9 (2019-12-22): Apply early patches from Debian
Closes #2
closed 2019-12-22 by antoineco