antoineco/broadcom-wl #4: Not working with Linux 4.13

chowbok, 2017-10-08:
I'm using the dkms version on Arch, and when I upgraded to linux 4.13.3, the networking stopped working. Same issue with linux 4.13.4. I have no problems when I go back down to 4.12.13.

I'm not sure if this is related, but I see these messages in the journal around the time it says the adapter is inactive:

DMAR: DRHD: handling fault status reg 2
DMAR: [DMA Read] Request device [02:00.0] fault addr fffeb000
 [fault reason 06] PTE Read access is not set

chowbok, 2017-10-08:
Actually, I think I figured this out. Looks like Arch enabled CONFIG_INTEL_IOMMU_DEFAULT_ON in the latest kernel config, which broke it on my system. Also looks like they backed out that change in 4.13.5, so I'll probably be okay once they release that. Sorry for the false alarm.

closed 2017-10-08 by chowbok

antoineco, 2017-10-08:
It might be useful information though, I haven't used 4.13 myself yet, but I'll keep your comment in mind if I run into trouble.