antoineco/broadcom-wl #5: After load wl.ko, no wl device found.

laowalter, 2018-01-13:
```
# lspci -vvnn |grep -n20 4331
466-            LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete-, EqualizationPhase1-
467-                     EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
468-    Capabilities: [100 v1] Advanced Error Reporting
469-            UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
470-            UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
471-            UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
472-            CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
473-            CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
474-            AERCap: First Error Pointer: 00, GenCap+ CGenEn- ChkCap+ ChkEn-
475-    Capabilities: [150 v1] Power Budgeting <?>
476-    Capabilities: [160 v1] Virtual Channel
477-            Caps:   LPEVC=0 RefClk=100ns PATEntryBits=1
478-            Arb:    Fixed- WRR32- WRR64- WRR128-
479-            Ctrl:   ArbSelect=Fixed
480-            Status: InProgress-
481-            VC0:    Caps:   PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
482-                    Arb:    Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
483-                    Ctrl:   Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
484-                    Status: NegoPending- InProgress-
485-
486:02:00.0 Network controller [0280]: Broadcom Limited BCM4331 802.11a/b/g/n [14e4:4331] (rev 02)
487:    Subsystem: Broadcom Limited BCM4331 802.11a/b/g/n [14e4:4331]
488-    Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
489-    Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
490-    Latency: 0, Cache Line Size: 256 bytes
491-    Interrupt: pin A routed to IRQ 17
492-    Region 0: Memory at a0600000 (64-bit, non-prefetchable) [size=16K]
493-    Capabilities: [40] Power Management version 3
494-            Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
495-            Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=2 PME-
496-    Capabilities: [58] Vendor Specific Information: Len=78 <?>
497-    Capabilities: [48] MSI: Enable- Count=1/1 Maskable- 64bit+
498-            Address: 0000000000000000  Data: 0000
499-    Capabilities: [d0] Express (v1) Endpoint, MSI 00
500-            DevCap: MaxPayload 128 bytes, PhantFunc 0, Latency L0s <4us, L1 unlimited
501-                    ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 10.000W
502-            DevCtl: Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
503-                    RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop-
504-                    MaxPayload 128 bytes, MaxReadReq 512 bytes
505-            DevSta: CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr+ TransPend-
506-            LnkCap: Port #0, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <2us, L1 <32us
507-                    ClockPM+ Surprise- LLActRep+ BwNot- ASPMOptComp-
```

```
# lsmod
Module                  Size  Used by
wl                   6340005  0
snd_intel_sst_pci       2425  0
snd_intel_sst_core     44202  1 snd_intel_sst_pci
snd_soc_sst_mfld_platform    67352  1 snd_intel_sst_core
lib80211_crypt_tkip     7458  0
lib80211                3378  1 lib80211_crypt_tkip
btusb                  28217  0
btrtl                   4640  1 btusb
btbcm                   6047  1 btusb
btintel                 6648  1 btusb
bluetooth             321848  5 btrtl,btintel,btbcm,btusb
firewire_ohci          30405  0
x86_pkg_temp_thermal     5509  0
firewire_core          48009  1 firewire_ohci
crc_itu_t               1531  1 firewire_core
```

```
# dmesg|grep -n10 wl

1011-[    3.576943] udevd[1895]: starting version 3.1.5
1012-[    3.653712] usb 2-1.8.1.3: New USB device found, idVendor=05ac, idProduct=828a
1013-[    3.653715] usb 2-1.8.1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
1014-[    3.653717] usb 2-1.8.1.3: Product: Bluetooth USB Host Controller
1015-[    3.653718] usb 2-1.8.1.3: Manufacturer: Apple Inc.
1016:[    3.682747] wl: loading out-of-tree module taints kernel.
1017:[    3.682750] wl: module license 'MIXED/Proprietary' taints kernel.
1018-[    3.682751] Disabling lock debugging due to kernel taint
1019-[    3.699414] Bluetooth: Core ver 2.22
1020-[    3.699424] NET: Registered protocol family 31
1021-[    3.699425] Bluetooth: HCI device and connection manager initialized
1022-[    3.699427] Bluetooth: HCI socket layer initialized
```

It seems no wl apter was drived.

antoineco, 2018-01-13:
`wl` is a module, not a device.

* `lsmod` shows that the module is loaded.
* `lspci`shows a broacom BCM433 adapter

What's the issue exactly?

closed 2018-02-13 by antoineco

arun-gurung, 2018-03-29:
wl.ko is the driver and the hardware is the device.

lsmod shows the wl driver
lspci shows the Broadcom Limited BCM4331 device