antoineco/broadcom-wl #6: Makefile does not compile on latest kernel

arun-gurung, 2018-03-29:
I updated to latest 4.15.14 rpm-package and now broadcom-wl will not compile. As I said, it compiles on 4.15.13.

This kernel comes from elrepo-kernel yum repository. http://elrepo.org/tiki/kernel-ml

antoineco, 2018-03-29:
Maybe similar to #4.

arun-gurung, 2018-03-31:
[output.txt](https://github.com/antoineco/broadcom-wl/files/1865772/output.txt)

arun-gurung, 2018-04-02:
I updated today to mainline kernel from el-repo and similar build error occurs.
[output.txt](https://github.com/antoineco/broadcom-wl/files/1867152/output.txt)

antoineco, 2018-04-04:
Are you sure you're building the correct branch? I just go this kernel today from my OS repo, and everything built just fine:

```
DKMS make.log for broadcom-wl-6.30.223.271 for kernel 4.15.14-300.fc27.x86_64 (x86_64)
Wed Apr  4 09:10:32 CEST 2018
Wireless Extension is the only possible API for this kernel version
Using Wireless Extension API
KBUILD_NOPEDANTIC=1 make -C /lib/modules/4.15.14-300.fc27.x86_64/build M=`pwd`
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/usr/src/kernels/4.15.14-300.fc27.x86_64'
CFG80211 API is prefered for this kernel version
Using CFG80211 API
  CC [M]  /var/lib/dkms/broadcom-wl/6.30.223.271/build/src/shared/linux_osl.o
  CC [M]  /var/lib/dkms/broadcom-wl/6.30.223.271/build/src/wl/sys/wl_linux.o
  CC [M]  /var/lib/dkms/broadcom-wl/6.30.223.271/build/src/wl/sys/wl_iw.o
  CC [M]  /var/lib/dkms/broadcom-wl/6.30.223.271/build/src/wl/sys/wl_cfg80211_hybrid.o
  LD [M]  /var/lib/dkms/broadcom-wl/6.30.223.271/build/wl.o
  Building modules, stage 2.
CFG80211 API is prefered for this kernel version
Using CFG80211 API
  MODPOST 1 modules
  CC      /var/lib/dkms/broadcom-wl/6.30.223.271/build/wl.mod.o
  LD [M]  /var/lib/dkms/broadcom-wl/6.30.223.271/build/wl.ko
make[1]: Leaving directory '/usr/src/kernels/4.15.14-300.fc27.x86_64'
```

arun-gurung, 2018-04-04:
I don't understand, because programming is over my head. I just installed the latest kernel from the repo and get the error.

antoineco, 2018-04-04:
This Git repository has two branches:
 * `master`: the original, non-patched driver sources from Broadcom
 * `patch-linux4.7`: (default branch) with all patches applied for Linux 4.7 up to Linux 4.15

Your build log suggests that you're currently on the `master` Git branch, according to the line numbers and function declarations, which is wrong. Please checkout `patch-linux4.7` and try again.

arun-gurung, 2018-04-04:
Thanks, you are right! I used the master instead of patch-linux4.7 and now it compiles.

closed 2018-04-04 by arun-gurung