antoineco/broadcom-wl #8: label:bug apple device 14e4:43ba

winehub, 2019-11-28:
lspci -nn | grep Network
03:00.0 Network controller [0280]: Broadcom Inc. and subsidiaries BCM43602 802.11ac Wireless LAN SoC [14e4:43ba] (rev 01)

this apple device works with brcmfmac and should be exclude form broadcom-wl.

Thanks for exclusion.

antoineco, 2019-11-28:
@winehub not sure if I understand your intention. Why are you using broadcom-wl if it doesn't support your device?

winehub, 2019-11-28:
@antoineco I have a moveable storage with linux on 2 seperated apple computers. Need always to include or exclude the driver. Because of the firmware I need a cold start.

winehub, 2019-11-28:
iMac 5K 2016 loads this driver, but not working.

antoineco, 2019-11-28:
Sounds to me like the module shouldn't be loaded in the first place. Loading a kernel module is a user action, and if you load the OS on heterogeneous hardware, the correct approach is to selectively load modules during boot, before the network comes up:
```bash
# in a systemd service of init script
if <detect your hardware here>; then
    modprobe -r wl
    modprobe brcmfmac
fi
```
You don't need a cold restart with that approach, modules can be loaded/unloaded dynamically, without blacklist.

Don't get me wrong, I understand the use case, but the code of this driver doesn't belong to me. I simply maintain a pre-patched codebase that works on the latest Linux kernel, the intention is not to fork the code and add features.

Are you building/installing the module via DKMS? In that case, I would suggest forking the repo and letting DKMS use your own fork. I can maybe assist you in excluding your specific driver if you need.

closed 2019-12-20 by antoineco