antoineco/broadcom-wl #10: Fails for gcc5+, kernel 5.4+

dcsouthwick, 2020-03-23:
Fails for modern kernels, modern gcc

```
/home/dsouthwi/broadcom-wl/src/wl/sys/wl_linux.c: In function ‘wl_attach’:
/home/dsouthwi/broadcom-wl/src/wl/sys/wl_linux.c:591:20: error: implicit declaration of function ‘ioremap_nocache’; did you mean ‘ioremap_cache’? [-Werror=implicit-function-declaration]
  591 |  if ((wl->regsva = ioremap_nocache(dev->base_addr, PCI_BAR0_WINSZ)) == NULL) {
      |                    ^~~~~~~~~~~~~~~
      |                    ioremap_cache
/home/dsouthwi/broadcom-wl/src/wl/sys/wl_linux.c:591:18: warning: assignment to ‘void *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
  591 |  if ((wl->regsva = ioremap_nocache(dev->base_addr, PCI_BAR0_WINSZ)) == NULL) {
      |                  ^
/home/dsouthwi/broadcom-wl/src/wl/sys/wl_linux.c: In function ‘wl_pci_probe’:
/home/dsouthwi/broadcom-wl/src/wl/sys/wl_linux.c:781:14: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
  781 |  bar1_addr = (uchar *)ioremap_nocache(pci_resource_start(pdev, 2), bar1_size);
      |              ^
/home/dsouthwi/broadcom-wl/src/wl/sys/wl_linux.c: In function ‘wl_reg_proc_entry’:
/home/dsouthwi/broadcom-wl/src/wl/sys/wl_linux.c:3377:58: error: passing argument 4 of ‘proc_create_data’ from incompatible pointer type [-Werror=incompatible-pointer-types]
 3377 |  if ((wl->proc_entry = proc_create_data(tmp, 0644, NULL, &wl_fops, wl)) == NULL) {
      |                                                          ^~~~~~~~
      |                                                          |
      |                                                          const struct file_operations *
In file included from /home/dsouthwi/broadcom-wl/src/wl/sys/wl_linux.c:38:
./include/linux/proc_fs.h:61:13: note: expected ‘const struct proc_ops *’ but argument is of type ‘const struct file_operations *’
   61 |             const struct proc_ops *,
      |             ^~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
```

antoineco, 2020-03-23:
Thanks for reporting! Let's see if we can take that patch and make it backwards compatible.

antoineco, 2020-03-31:
@dcsouthwick I pushed a patch in 5633ca046506b71b63ef40938e7cb8e54ddeefca based on the bug report you shared, but currently I don't have access to the machine I own that has this network adapter. Would it be possible for you to try the code on the [`fix-kernel-5.4`](https://github.com/antoineco/broadcom-wl/tree/fix-kernel-5.4) branch?

```
$ git fetch
$ git checkout fix-kernel-5.4
$ sudo make
$ sudo make install
$ sudo depmod -A
$ sudo modprobe wl
```

antoineco, 2020-04-02:
Closed by 5633ca046506b71b63ef40938e7cb8e54ddeefca

closed 2020-04-02 by antoineco