antoineco/broadcom-wl #13: broadcom BCM4360 firmware

joker5bb, 2020-04-21:
I have the 
03:00.0 Network controller [0280]: Broadcom Inc. and subsidiaries BCM4360 802.11ac Wireless Network Adapter [14e4:43a0] (rev 03)

The wl driver works, but Im trying to upgrade to the open source brcmfmac driver and it requires firmware. How does this driver work without firmware?

antoineco, 2020-04-22:
Hi @joker5bb, first of all beware that the brcmfmac driver may not support the BCM4360 chip. To my knowledge this driver supports BCM43602, which is a different chip despite the very similar name. Last time I checked, only `wl` supported BCM4360.

Regarding the firmware, that's an interesting question. A firmware is usually a piece of software loaded in a device's memory. Linux only needs an extra _firmware file_ when the driver can not handle the hardware on its own, which is not the case of the proprietary `wl` driver.

I found this very detailed answer on linuxquestions org:
https://www.linuxquestions.org/questions/linux-newbie-8/files-in-lib-firmware-623708/#post3069146

antoineco, 2020-04-29:
@joker5bb did you manage to make  the `brcmfmac` driver work with your BCM4360 chip?

newperson1746, 2020-05-03:
check out https://github.com/MerlinRdev/86u-merlin/tree/master/release/src-rt-5.02hnd/bcmdrivers/broadcom/net/wl/impl51/4365/src/wl

seems to have full src for wl

antoineco, 2020-05-13:
I'm not sure if I understand exactly what the expectation is here, but I do hope I answered the original question. If not, feel free to re-open!

closed 2020-05-13 by antoineco

satmandu, 2021-08-10:
Sorry to necrobump, but @joker5bb did you ever get a working brcmfmac driver for that card? Another ID I'm seeing in Broadcom header files for this card appears to be BCM4360_D11AC_ID, for instance here: https://github.com/RMerl/asuswrt-merlin.ng/blob/6b60627c8c9c5c0271e956c914afb5277f81f9c4/release/src-rt-6.x.4708/include/bcmdevs.h

Did the Cypress acquisition of Broadcom's IoT IP get them the info on this chip and the wl interface driver?

I guess the softmac wl support for this chipset is going to remain b broken unless we figure out how to port over the wl software from the other BCM4360 wl drivers like at https://github.com/RMerl/asuswrt-merlin.ng ?