antoineco/broadcom-wl #15: i followed the instructions to automatically add wl via dkms but not working

bukyapavankumar, 2020-08-19:
sudo dkms status
broadcom-wl, 6.30.223.271: added
nvidia, 390.138, 5.7.15-977.native, x86_64: installed

antoineco, 2020-08-19:
> Providing that the dkms service is enabled, the module should appear as installed in the list of modules managed by DKMS
> \>> **after the system boots for the first time on the new kernel** <<
>
> https://github.com/antoineco/broadcom-wl#automatically

bukyapavankumar, 2020-08-19:
> > Providing that the dkms service is enabled, the module should appear as installed in the list of modules managed by DKMS
> > >> **after the system boots for the first time on the new kernel** <<
> > https://github.com/antoineco/broadcom-wl#automatically

lspci -k doesnt show wl module ,but dkms status show

antoineco, 2020-08-19:
That's expected if you didn't reboot. Your module is **added**, not **installed**.

bukyapavankumar, 2020-08-19:
i did reboot

antoineco, 2020-08-19:
Could you please show the output of those 3 commands?
```console
$ sudo modprobe wl
```
```console
$ sudo modprobe wl
```
```console
$ dmesg
```

bukyapavankumar, 2020-08-19:
wl module not found

antoineco, 2020-08-19:
> the output of those 3 commands

antoineco, 2020-08-19:
And maybe also `dkms install broadcom-wl/6.30.223.271` if possible.

bukyapavankumar, 2020-08-19:
> > the output of those 3 commands

1. nothing
2.modprobe: FATAL: Module wl not found in directory /lib/modules/5.7.15-977.native

bukyapavankumar, 2020-08-19:
> And maybe also `sudo dkms install broadcom-wl/6.30.223.271` if possible.
it works after this command.
dkms doesnt install it automatically?

bukyapavankumar, 2020-08-19:
mobile hotspot doesnt work though

antoineco, 2020-08-19:
Nice! That's an interesting behaviour, it should do it automatically because this option is defined:
https://github.com/antoineco/broadcom-wl/blob/a9362f4e4416d8e842b976fd243445711363ec24/dkms.conf#L7

May I ask what's the `dkms` version you're using? Maybe it requires an update to the `dkms.conf`.

bukyapavankumar, 2020-08-19:
> Nice! That's an interesting behaviour, it should do it automatically because this option is defined: [...]

dkms2.8

antoineco, 2020-08-19:
OK thanks. Maybe [`dkms_autoinstaller`](http://manpages.ubuntu.com/manpages/xenial/man8/dkms.8.html#dkms_autoinstaller) simply doesn't run on boot on your system. `systemctl status dkms` should tell you whether or not the service is enabled.

Regarding the mobile hotspot, unfortunately I only backport patches so the module can build and run on modern Linux kernels. Troubleshooting issues with specific hardware would require a larger investigation.

bukyapavankumar, 2020-08-19:
> systemctl status dkms

Unit dkms.service could not be found.

bukyapavankumar, 2020-08-19:
> OK thanks. Maybe [`dkms_autoinstaller`](http://manpages.ubuntu.com/manpages/xenial/man8/dkms.8.html#dkms_autoinstaller) simply doesn't run on boot on your system. `systemctl status dkms` should tell [...]

actually mobile hotspot works but in wep .linux driver doesnt have wpa support whereas windows driver has it.anyway can we run/modify the driver to make it work?

antoineco, 2020-08-19:
Hmm, `wl` does support WPA/WPA2. I'm running this module on a Linux 5.6 kernel and my router (Linksys WRT1900AC) uses WPA2. I also occasionally connect to an Android 9 hotspot that is also using WPA2.

Some hardware may be supported better than some other, like with any driver. The purpose of this repo is not to provide a fork of `wl`, I wouldn't include patches that aren't purely related to kernel compatibility, besides I'm not a kernel developer myself so fixing such issue would be quite challenging for me.

bukyapavankumar, 2020-08-19:
> Hmm, `wl` does support WPA/WPA2. I'm running this module on a Linux 5.6 kernel and my router (Linksys WRT1900AC) uses WPA2. I also occasionally connect to an Android 9 hotspot (Sony Xperia Z3 Compact) that is also using WPA2.
> 
> Some hardware may be supported better than some other, like with any driver. The purpose of this repo is not to provide a fork of `wl`, I wouldn't include patches that aren't purely related to kernel compatibility, besides I'm not a kernel developer myself so fixing such issue would be quite challenging for me.

I mean it doesn't support creation of WPA/wpa2 supported hotspot .it can create only wep supported hotspot.

bukyapavankumar, 2020-08-19:
iw list
Wiphy phy0
	max # scan SSIDs: 1
	max scan IEs length: 0 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CMAC (00-0f-ac:6)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
		 * managed
	Band 1:
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps (short preamble supported)
			* 5.5 Mbps (short preamble supported)
			* 11.0 Mbps (short preamble supported)
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412 MHz [1] (20.0 dBm)
			* 2417 MHz [2] (20.0 dBm)
			* 2422 MHz [3] (20.0 dBm)
			* 2427 MHz [4] (20.0 dBm)
			* 2432 MHz [5] (20.0 dBm)
			* 2437 MHz [6] (20.0 dBm)
			* 2442 MHz [7] (20.0 dBm)
			* 2447 MHz [8] (20.0 dBm)
			* 2452 MHz [9] (20.0 dBm)
			* 2457 MHz [10] (20.0 dBm)
			* 2462 MHz [11] (20.0 dBm)
			* 2467 MHz [12] (20.0 dBm) (no IR)
			* 2472 MHz [13] (20.0 dBm) (no IR)
			* 2484 MHz [14] (20.0 dBm) (no IR)
	Band 2:
		Bitrates (non-HT):
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 5170 MHz [34] (disabled)
			* 5180 MHz [36] (20.0 dBm) (no IR)
			* 5190 MHz [38] (20.0 dBm) (no IR)
			* 5200 MHz [40] (20.0 dBm) (no IR)
			* 5210 MHz [42] (20.0 dBm) (no IR)
			* 5220 MHz [44] (20.0 dBm) (no IR)
			* 5230 MHz [46] (20.0 dBm) (no IR)
			* 5240 MHz [48] (20.0 dBm) (no IR)
			* 5260 MHz [52] (20.0 dBm) (no IR, radar detection)
			* 5280 MHz [56] (20.0 dBm) (no IR, radar detection)
			* 5300 MHz [60] (20.0 dBm) (no IR, radar detection)
			* 5320 MHz [64] (20.0 dBm) (no IR, radar detection)
			* 5500 MHz [100] (20.0 dBm) (no IR, radar detection)
			* 5520 MHz [104] (20.0 dBm) (no IR, radar detection)
			* 5540 MHz [108] (20.0 dBm) (no IR, radar detection)
			* 5560 MHz [112] (20.0 dBm) (no IR, radar detection)
			* 5580 MHz [116] (20.0 dBm) (no IR, radar detection)
			* 5600 MHz [120] (20.0 dBm) (no IR, radar detection)
			* 5620 MHz [124] (20.0 dBm) (no IR, radar detection)
			* 5640 MHz [128] (20.0 dBm) (no IR, radar detection)
			* 5660 MHz [132] (20.0 dBm) (no IR, radar detection)
			* 5680 MHz [136] (20.0 dBm) (no IR, radar detection)
			* 5700 MHz [140] (20.0 dBm) (no IR, radar detection)
			* 5745 MHz [149] (20.0 dBm) (no IR)
			* 5765 MHz [153] (20.0 dBm) (no IR)
			* 5785 MHz [157] (20.0 dBm) (no IR)
			* 5805 MHz [161] (20.0 dBm) (no IR)
			* 5825 MHz [165] (20.0 dBm) (no IR)
			* 5920 MHz [184] (disabled)
			* 5940 MHz [188] (disabled)
			* 5960 MHz [192] (disabled)
			* 5980 MHz [196] (disabled)
			* 6000 MHz [200] (disabled)
			* 6020 MHz [204] (disabled)
			* 6040 MHz [208] (disabled)
			* 6060 MHz [212] (disabled)
			* 6080 MHz [216] (disabled)
	Supported commands:
		 * set_interface
		 * new_key
		 * join_ibss
		 * set_pmksa
		 * del_pmksa
		 * flush_pmksa
		 * connect
		 * disconnect
	software interface modes (can always be added):
	interface combinations are not supported
	Device supports scan flush.
	Supported extended features:

bukyapavankumar, 2020-08-19:
anyways closing this

closed 2020-08-19 by bukyapavankumar

antoineco, 2020-08-19:
@bukyapavankumar oh, my bad! Yes you're right about that. I wish Broadcom could support `wl` but unfortunately it seems like it's been completely abandoned.