antoineco/broadcom-wl #16: Build error -Wimplicit-function-declaration with 5.9-rc4

dilyn-corner, 2020-09-08:
Just looking towards the future here with this one. If I had any level of C knowledge, I would provide a patch. Based on my knowledge, the fix should be easy though! 

In `src/wl/sys/wl_linux.c`, `segment_eq(get_fs(), KERNEL_DS)` is implicitly declared. It seems this is the only such function.

antoineco, 2020-09-08:
Thanks for reporting, @dilyn-corner! I haven't tried that kernel yet, but judging by the error I think you found the culprit.

I'll look into it.

antoineco, 2020-09-10:
I'm not entirely sure what happened yet, but [this patch](https://lore.kernel.org/linux-arch/20200710135706.537715-5-hch@lst.de/) may have removed the declaration of `segment_eq` from `asm/uaccess.h`, which is included here:

https://github.com/antoineco/broadcom-wl/blob/a9362f4e4416d8e842b976fd243445711363ec24/src/wl/sys/wl_linux.c#L58

antoineco, 2020-09-10:
Someone was actually faster to fix it :)

https://gist.github.com/joanbm/9e20b3842bb336379eccc12410901e37

I'm committing this right now. Thanks again 🙌

commit fd32a8e9 (2020-09-10): Fix implicit declaration of segment_eq in Linux 5.9

Source: https://gist.github.com/joanbm/9e20b3842bb336379eccc12410901e37

See also: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=428e2976a5bf7e7f5554286d7a5a33b8147b106a

Fixes #16
closed 2020-09-10 by antoineco