antoineco/broadcom-wl #17: build error on kernel 5.10

rainerklier, 2020-12-14:
trying to build driver in kernel 5.10 results in:
```
KBUILD_NOPEDANTIC=1 make -C /lib/modules/`uname -r`/build M=`pwd`
make[1]: Entering directory '/usr/src/linux-5.10.0-1.g1bdd4f9-obj/x86_64/default'
CFG80211 API is prefered for this kernel version
Using CFG80211 API
  CC [M]  /usr/src/packages/SOURCES/broadcom-wl-patch-linux4.7/src/shared/linux_osl.o
  CC [M]  /usr/src/packages/SOURCES/broadcom-wl-patch-linux4.7/src/wl/sys/wl_linux.o
  CC [M]  /usr/src/packages/SOURCES/broadcom-wl-patch-linux4.7/src/wl/sys/wl_iw.o
  CC [M]  /usr/src/packages/SOURCES/broadcom-wl-patch-linux4.7/src/wl/sys/wl_cfg80211_hybrid.o
/usr/src/packages/SOURCES/broadcom-wl-patch-linux4.7/src/wl/sys/wl_cfg80211_hybrid.c: In function ‘wl_dev_ioctl’:
/usr/src/packages/SOURCES/broadcom-wl-patch-linux4.7/src/wl/sys/wl_cfg80211_hybrid.c:459:7: error: implicit declaration of function ‘get_fs’; did you mean ‘sget_fc’? [-Werror=implicit-function-declaration]
  459 |  fs = get_fs();
      |       ^~~~~~
      |       sget_fc
/usr/src/packages/SOURCES/broadcom-wl-patch-linux4.7/src/wl/sys/wl_cfg80211_hybrid.c:459:7: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
/usr/src/packages/SOURCES/broadcom-wl-patch-linux4.7/src/wl/sys/wl_cfg80211_hybrid.c:461:2: error: implicit declaration of function ‘set_fs’; did you mean ‘sget_fc’? [-Werror=implicit-function-declaration]
  461 |  set_fs(KERNEL_DS);
      |  ^~~~~~
      |  sget_fc
/usr/src/packages/SOURCES/broadcom-wl-patch-linux4.7/src/wl/sys/wl_cfg80211_hybrid.c:461:9: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
  461 |  set_fs(KERNEL_DS);
      |         ^~~~~~~~~
      |         KERNFS_NS
/usr/src/packages/SOURCES/broadcom-wl-patch-linux4.7/src/wl/sys/wl_cfg80211_hybrid.c:461:9: note: each undeclared identifier is reported only once for each function it appears in
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-5.10.0-1.g1bdd4f9/scripts/Makefile.build:280: /usr/src/packages/SOURCES/broadcom-wl-patch-linux4.7/src/wl/sys/wl_cfg80211_hybrid.o] Error 1
make[2]: *** [/usr/src/linux-5.10.0-1.g1bdd4f9/Makefile:1819: /usr/src/packages/SOURCES/broadcom-wl-patch-linux4.7] Error 2
make[1]: *** [../../../linux-5.10.0-1.g1bdd4f9/Makefile:185: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-5.10.0-1.g1bdd4f9-obj/x86_64/default'
make: *** [Makefile:159: all] Error 2
```

this is because of function `set_fs()` is not any more available in kernel 5.10

antoineco, 2020-12-14:
Thanks for reporting 👍  That one is definitely tricky. I've seen 2 attempts to fix it already:

* https://gist.github.com/joanbm/5c640ac074d27fd1d82c74a5b67a1290
* https://pkgs.rpmfusion.org/cgit/nonfree/wl-kmod.git/commit/?id=d18601cec354466a11aae2e9598121519fd8c74e

Let's see if they work.

rainerklier, 2020-12-14:
yes.
i think this is the same what i found:
https://lists.links2linux.de/pipermail/packman/2020-November/016479.html
https://pmbs.links2linux.org/package/show/home:manfred.h/broadcom-wl
https://pmbs.links2linux.org/package/view_file/Essentials/broadcom-wl/broadcom-wl.linux-5.9.patch?rev=ad63dd95e33c6cec99504e37af076606

rainerklier, 2020-12-14:
i tried the first in your list, which seems to be the same as the one i found (packman/links2linux), and it seems to work.

antoineco, 2020-12-14:
Right, the second one allows the module to build on 5.10, but not to run, unless I'm missing the point.

antoineco, 2020-12-14:
I have concerns with the first patch actually, because it effectively rewrites part of the code. This is likely to break backward compatibility and I'd prefer to avoid that. 

According to the commit message in https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/arch/x86/include/asm/uaccess.h?id=47058bb54b57962b3958a936ddbc59355e4c5504, using `set_fs()` might actually not be needed at all anymore. I think we can execute `do_ioctl` straight, like the second patch did.

antoineco, 2020-12-14:
@rainerklier I have a (potential) fix ready in #18. Do you have cycles to test it?

rainerklier, 2020-12-14:
> @rainerklier I have a (potential) fix ready in #18. Do you have cycles to test it?

yes.
how/where can i download the source to try to build and run it?

antoineco, 2020-12-14:
You can download the Zip or clone the repo from https://github.com/antoineco/broadcom-wl/tree/linux-5.10 (notice the branch, "5.10").

rainerklier, 2020-12-14:
ok, i will try

rainerklier, 2020-12-14:
ok, i was able to compile and install the module.
but when the module loaded the following errors came up in dmesg:
```[   19.005738] ERROR @wl_dev_intvar_get : 
[   19.005740] error (-14)
[   19.005741] ERROR @wl_cfg80211_get_tx_power : 
[   19.005741] error (-14)
[   19.005760] ERROR @wl_dev_intvar_get : 
[   19.005761] error (-14)
[   19.005762] ERROR @wl_cfg80211_get_tx_power : 
[   19.005762] error (-14)
[   19.005885] ERROR @wl_dev_intvar_get : 
[   19.005886] error (-14)
[   19.005887] ERROR @wl_cfg80211_get_tx_power : 
[   19.005888] error (-14)
[   19.005897] ERROR @wl_dev_intvar_get : 
[   19.005898] error (-14)
[   19.005899] ERROR @wl_cfg80211_get_tx_power : 
[   19.005900] error (-14)
[   19.006104] ERROR @wl_dev_intvar_get : 
[   19.006105] error (-14)
[   19.006106] ERROR @wl_cfg80211_get_tx_power : 
[   19.006106] error (-14)
[   29.006637] net_ratelimit: 6 callbacks suppressed
[   29.006638] ERROR @wl_dev_intvar_get : 
[   29.006640] error (-14)
[   29.006641] ERROR @wl_cfg80211_get_tx_power : 
[   29.006642] error (-14)
[   29.006661] ERROR @wl_dev_intvar_get : 
```

antoineco, 2020-12-14:
My bad, I forgot to patch `src/wl/sys/wl_linux.c`, which also calls `get_fs()`. I just pushed another commit to the branch. Hopefully that is enough to fix the issue.

rainerklier, 2020-12-14:
i will give it a try.

rainerklier, 2020-12-14:
i saw the difference in file src/wl/sys/wl_linux.c, but the result is the same, when loading the module:
```[   19.005890] net_ratelimit: 16 callbacks suppressed
[   19.005891] ERROR @wl_dev_intvar_get : 
[   19.005892] error (-14)
[   19.005893] ERROR @wl_cfg80211_get_tx_power : 
[   19.005893] error (-14)
[   19.005907] ERROR @wl_dev_intvar_get : 
[   19.005907] error (-14)
[   19.005907] ERROR @wl_cfg80211_get_tx_power : 
[   19.005908] error (-14)
[   19.005987] ERROR @wl_dev_intvar_get : 
[   19.005987] error (-14)
[   19.005988] ERROR @wl_cfg80211_get_tx_power : 
[   19.005988] error (-14)
[   19.005994] ERROR @wl_dev_intvar_get : 
[   19.005994] error (-14)
[   19.005995] ERROR @wl_cfg80211_get_tx_power : 
[   19.005995] error (-14)
[   19.006115] ERROR @wl_dev_intvar_get : 
[   19.006116] error (-14)
[   19.006116] ERROR @wl_cfg80211_get_tx_power : 
[   19.006117] error (-14)
[   29.005851] net_ratelimit: 6 callbacks suppressed
[   29.005852] ERROR @wl_dev_intvar_get : 
[   29.005854] error (-14)
[   29.005855] ERROR @wl_cfg80211_get_tx_power : 
[   29.005856] error (-14)
[   29.005884] ERROR @wl_dev_intvar_get : 
[   29.005885] error (-14)
[   29.005886] ERROR @wl_cfg80211_get_tx_power : 
[   29.005887] error (-14)
[   29.006013] ERROR @wl_dev_intvar_get : 
[   29.006014] error (-14)
[   29.006015] ERROR @wl_cfg80211_get_tx_power : 
[   29.006015] error (-14)
[   29.006028] ERROR @wl_dev_intvar_get : 
[   29.006029] error (-14)
[   29.006030] ERROR @wl_cfg80211_get_tx_power : 
[   29.006030] error (-14)
[   29.006221] ERROR @wl_dev_intvar_get : 
[   29.006222] error (-14)
[   29.006222] ERROR @wl_cfg80211_get_tx_power : 
[   29.006223] error (-14)
[   38.148883] net_ratelimit: 6 callbacks suppressed
[   38.148885] ERROR @wl_cfg80211_set_power_mgmt : 
[   38.148886] error (-14)
[   38.153225] ERROR @wl_cfg80211_set_power_mgmt : 
[   38.153226] error (-14)```

antoineco, 2020-12-14:
@rainerklier thanks a lot for spending the time to test it. Looks like I will have to take another route.

rainerklier, 2020-12-14:
> @rainerklier thanks a lot for spending the time to test it. Looks like I will have to take another route.

no problem.

meanwhile i use a module which is created with the first patch.
i found out, that this patch was already used to create a module package for kernel 5.9.

i am in touch with the maintainer of https://build.opensuse.org/package/show/home:manfred-h:KMPs/broadcom-wl who creates the repository http://pmbs.links2linux.de:8080/Essentials/Factory/
about a month ago he created the wl module with the patch: https://build.opensuse.org/package/view_file/home:manfred-h:KMPs/broadcom-wl/broadcom-wl.linux-5.9.patch?expand=1
at that time, this patch was targeting kernel 5.9, but it is also a fix for kernel 5.10.

rekeeszeng, 2020-12-17:
@antoineco  I tried branch 5.10 yesterday. The Wi-Fi adapter could not scan for networks. After changing the two files in #18, the same problem persisted. Branch 5.10 is gone now? Any suggestion how I can get the driver working?

rainerklier, 2020-12-17:
> @antoineco I tried branch 5.10 yesterday. The Wi-Fi adapter could not scan for networks. After changing the two files in #18, the same problem persisted. Branch 5.10 is gone now? Any suggestion on how I can get the driver working? Thanks.

for me, the patch shown under https://build.opensuse.org/package/view_file/home:manfred-h:KMPs/broadcom-wl/broadcom-wl.linux-5.9.patch?expand=1 works

first, i tried it on the sources on my own, and it worked.
now, i use the already pre-compiled module from repo, which ist more easy. ;-)

antoineco, 2020-12-17:
@rainerklier yes the branch is gone because that patch wasn't enough to get the driver to work.

If the other patch works for you and you already found a package that includes it I'd suggest using that until a find a suitable approach without breaking backwards compatibility.

rekeeszeng, 2020-12-18:
> for me, the patch shown under https://build.opensuse.org/package/view_file/home:manfred-h:KMPs/broadcom-wl/broadcom-wl.linux-5.9.patch?expand=1 works
>
> first, i tried it on the sources on my own, and [...]

Is this only for OpenSUSE? Can I apply it to a Debian-based distro?

rainerklier, 2020-12-18:
> > first, i tried it on the sources on my own, and it worked.
> > now, i use the already pre-compiled module from repo, which ist more easy. ;-)
> 
> This repo seems to be only for OpenSUSE? I'm on a Debian-based distro. I'll try the source. Thanks.

yes, the repo is for opensuse.
but, as i wrote, before using the repo, i manually changed the source files following the patch and it worked.
and this will work on any distro.
yes, just download the source code from here, and manually do the patch and build module manually.

rekeeszeng, 2020-12-18:
> yes, the repo is for opensuse.
> but, as i wrote, before using the repo, i manually changed the source files following the patch and it worked.
> and this will work on any distro.
> yes, just download the [...]

I made all the changes described in the patch for kernel 5.9 and then installed kernel 5.10. It worked! Thanks a lot.

loopingstrategy, 2020-12-29:
Hello, 

I used Ubuntu 20.10 with kernel 5.10.3

I tried to apply the patch https://build.opensuse.org/package/view_file/home:manfred-h:KMPs/broadcom-wl/broadcom-wl.linux-5.9.patch?expand=1 but i had the following error  : 

` *** No rule to make target 'scripts/module.lds', needed by`

And i found a  quick fix https://github.com/Mange/rtl8192eu-linux-driver/issues/205#issuecomment-736366347 

If it can help someone, thank you for this exchange, for this repository and all your work !

SukkoPera, 2021-01-11:
I also applied [this patch](https://gist.github.com/joanbm/5c640ac074d27fd1d82c74a5b67a1290) manually and it seems to be working fine for me, too.

I'm attaching the diff wrt current HEAD, hopefully more people can test it.

Thanks for this repo, btw!

[broadcom-wl-fix-linux-5.10.patch.txt](https://github.com/antoineco/broadcom-wl/files/5797120/broadcom-wl-fix-linux-5.10.patch.txt)

antoineco, 2021-03-11:
Thank you everyone for your patience, and sorry for the delay! I have backported the patch in a backward-compatible way (all credits go to the original author and to @NVieville who contributed this to RPM Fusion).

commit 34633bb3 (2021-03-11): Get rid of get_fs/set_fs calls in Linux >= 5.10

Backported from RPM Fusion:
https://pkgs.rpmfusion.org/cgit/nonfree/wl-kmod.git/commit/?id=f46d7c73586ceac4e9f23b572f5cb10a78abfa96

Original patch by Joan Bruguera:
https://gist.github.com/joanbm/5c640ac074d27fd1d82c74a5b67a1290

Fixes #17
closed 2021-03-11 by antoineco