antoineco/broadcom-wl #21: Higher 5ghz channels?

satmandu, 2021-07-16:
Is there a way to get higher 5Ghz channels working with this driver?

All I see are lower channels below 100:
```
iwlist chan   
lo        no frequency information.

wlp2s0    32 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 32 : 5.16 GHz
          Channel 34 : 5.17 GHz
          Channel 36 : 5.18 GHz
          Channel 38 : 5.19 GHz
          Channel 40 : 5.2 GHz
          Channel 42 : 5.21 GHz
          Channel 44 : 5.22 GHz
          Channel 46 : 5.23 GHz
          Channel 48 : 5.24 GHz
          Channel 50 : 5.25 GHz
          Channel 52 : 5.26 GHz
          Channel 54 : 5.27 GHz
          Channel 56 : 5.28 GHz
          Channel 58 : 5.29 GHz
          Channel 60 : 5.3 GHz
          Channel 62 : 5.31 GHz
          Channel 64 : 5.32 GHz
          Channel 66 : 5.33 GHz
          Channel 68 : 5.34 GHz
          Channel 96 : 5.48 GHz
          Channel 98 : 5.49 GHz
          Current Frequency:2.447 GHz (Channel 8)
```

antoineco, 2021-07-19:
@satmandu I believe it should be able to support up to 224 channels, by looking at this constant which is used in a few places: 

https://github.com/antoineco/broadcom-wl/blob/34633bb3c243eb060d33b144836d05b143517892/src/shared/bcmwifi/include/bcmwifi_channels.h#L44

I assume you've ensured the access point supports higher frequencies, based on your geographical location?

satmandu, 2021-07-19:
Yes. I will try rebuilding...

satmandu, 2021-08-09:
Rebuilt.
No difference.
```
ubuntu@ubuntu:~/src/broadcom-wl$ iw reg get
global
country US: DFS-FCC
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 23), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)

ubuntu@ubuntu:~/src/broadcom-wl$ iwlist chan   
lo        no frequency information.

enx806d970aa85a  no frequency information.

wlp2s0    32 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 32 : 5.16 GHz
          Channel 34 : 5.17 GHz
          Channel 36 : 5.18 GHz
          Channel 38 : 5.19 GHz
          Channel 40 : 5.2 GHz
          Channel 42 : 5.21 GHz
          Channel 44 : 5.22 GHz
          Channel 46 : 5.23 GHz
          Channel 48 : 5.24 GHz
          Channel 50 : 5.25 GHz
          Channel 52 : 5.26 GHz
          Channel 54 : 5.27 GHz
          Channel 56 : 5.28 GHz
          Channel 58 : 5.29 GHz
          Channel 60 : 5.3 GHz
          Channel 62 : 5.31 GHz
          Channel 64 : 5.32 GHz
          Channel 66 : 5.33 GHz
          Channel 68 : 5.34 GHz
          Channel 96 : 5.48 GHz
          Channel 98 : 5.49 GHz
```

Any suggestions for debug output to help figure out where this is going wrong?

The SSID is broadcasting on channel 149, and is visible when I reboot into macos.

satmandu, 2021-08-09:
More info:
dmesg:
```
[ 1868.624716] wlan0: Broadcom BCM43a0 802.11 Hybrid Wireless Controller 6.30.223.271 (r587334)
```
lspci:
```
02:00.0 Network controller: Broadcom Inc. and subsidiaries BCM4360 802.11ac Wireless Network Adapter (rev 03)
```

satmandu, 2021-08-09:
I wonder if this is a firmware issue which can be addressed by newer firmware we might be able to extract from newer apple drivers? e.g. the drivers in https://support.apple.com/en-us/HT209600 as mentioned here:

https://blog.quarkslab.com/reverse-engineering-broadcom-wireless-chipsets.html

antoineco, 2021-08-10:
Maybe, but the aim of this repo is simply to patch `wl` for recent Linux kernels. There is no active feature enhancement or bug fixing of Broadcom's proprietary code happening here.

The BCM4360 chip was always somewhat problematic because it isn't supported by open-source alternatives to Broadcom's drivers like [`brcm80211`](https://wireless.wiki.kernel.org/en/users/Drivers/brcm80211). I heard it's because the firmware itself is closed source but yeah, it doesn't explain why you see higher channels on one OS and not the other.

antoineco, 2021-08-10:
Very interesting article btw! Long read but good read, a lot of previous information about those firmwares.

satmandu, 2021-08-10:
I really hope we can get these firmwares patched to fix these issues. Insane that this fairly recent hardware has such broken linux drivers.

antoineco, 2022-03-11:
Closing because I feel like everything has been discussed, and the issue is related to the core features of the module itself, which I'm not the author of and am not allowed to modify based on the software license.

closed 2022-03-11 by antoineco