antoineco/broadcom-wl #23: Connection bitrate show up as 0 kb/s

wreckdump, 2022-01-04:
I am using BCM43228 card, the connection is fine and it works, but the connection bitrate shows up as 0 kb/s. With an Intel card, it worked fine. My setup is like the following

OS : Arch Linux (5.15.12)
Network manager : wpa_supplicant
Windows manager : i3 WM with i3 status bar

i3status bar queries nl80211 driver output, I think. So, I tried forcing wpa_supplicant to use nl80211 driver, but nothing changed (network connection works, but still 0 kb/s for the bitrate).

antoineco, 2022-01-04:
@wreckdump there won't be any feature or bug fix implemented in this repository. The code is the property of Broadcom and I simply apply patches that make it work on recent versions of the Linux kernel.

With that being said, I see your card is a BCM43228. Wouldn't the [b43 driver](https://wireless.wiki.kernel.org/en/users/Drivers/b43#Supported_devices) be more suitable for that chip?

wreckdump, 2022-01-04:
Ahh! I was under the impression that `broadcom-wl` would work fine. I'll try `b34`.Thanks!

closed 2022-01-04 by wreckdump