antoineco/broadcom-wl #24: Problem with BCM43142

Andrea055, 2022-04-29:
Hi. I have problem with BCM43142 and HP x360 i3-4010T on Ubuntu 16,18,20 and 22.
The card result hard blocked in rfkill but there are no buttons.
When i try to up the interface up the wl driver return error -1 in dmesg.
Any solution?

Andrea055, 2022-04-29:
Already, tried but wifi not Is reconized

antoineco, 2022-04-29:
According to [this Wiki](https://wiki.ubuntuusers.de/WLAN/Karten/Broadcom) (in German) you need this additional Broadcom Bluetooth firmware: https://github.com/winterheart/broadcom-bt-firmware

_tl;dr;_

```console
$ dmesg | grep -i bluetooth | grep -i firmware
bluetooth hci1: Direct firmware load for brcm/BCM20702A1-0b05-17cb.hcd failed with error -2
```

Take note of the expected `.hcd` file name. In this example, it is `BCM20702A1-0b05-17cb.hcd`.
Then copy it from the aforementioned repository:

```
git clone https://github.com/winterheart/broadcom-bt-firmware
cd ~/broadcom-bt-firmware/brcm
sudo cp BCM20702A1-0b05-17cb.hcd /lib/firmware/brcm/
sudo modprobe -rfv btusb
sudo modprobe -v btusb
dmesg | grep -i blue
hciconfig --all
```

Unfortunately I don't have this device to try myself.

Andrea055, 2022-05-04:
I install this  but not work, any other solutions?

antoineco, 2022-05-04:
Unfortunately I don't know. This chip doesn't seem to be well supported on Linux due to the dual Bluetooth+WiFi functionalities, if I understood everything correctly.

Most of the workarounds I see on the internet rely on files extracted from Windows drivers.

Still, the link I shared with you seems reliable. Could you please post the output of all the commands I wrote above? Maybe there is something I can help with. No guarantee, but worth a try!

antoineco, 2022-06-05:
Closing due to inactivity, and because the problem seems to be related to the Bluetooth functionality of this device.

closed 2022-06-05 by antoineco