antoineco/broadcom-wl #25: more source?

djdisodo, 2022-05-09:
[repo](https://github.com/Jackysi/advancedtomato/tree/advancedtomato-ac/release/src-rt)
it's not open source tho we can get sources

this driver may also support ap mode
which can be used on routers

antoineco, 2022-05-09:
Could you please be more specific about the request?

I do see a `wl/` directory at the link you shared, but it contains the version `v5.110.27.20012` of the `wl` driver, while this repository is based on `v6.30.223.271`.

djdisodo, 2022-05-09:
@antoineco can you check [here](https://github.com/Jackysi/advancedtomato/tree/advancedtomato-ac/release/src-rt-6.x/wl/linux)?

i want to get this worked on my router,
i couldn't get any wl drivers for arm that works on current kernel
but this link seems to include arm blobs, and some more for ap mode

do you have any idea i can use these blobs to compile kernel mod for current kernel?

antoineco, 2022-05-09:
The repository doesn't seem to contain any information about the driver, it's all very opaque. I can't even tell whether the artifacts have anything to do with the original `wl` driver since the source isn't available.
I'm afraid I don't know more than you unfortunately.

djdisodo, 2022-05-09:
```
sodo@DESKTOP-S799JDR:~/broadcom-wl$ cat Makefile | tail -n 15

# Cross compile setup.  Tool chain and kernel tree, replace with your own.
CROSS_TOOLS        = /home/sodo/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-11.2.0_musl_eabi/bin/arm-openwrt-linux-
CROSS_KBUILD_DIR   = /home/sodo/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/linux-5.10.113/
all:
        KBUILD_NOPEDANTIC=1 make -C $(KBUILD_DIR) M=`pwd`

cross:
        KBUILD_NOPEDANTIC=1 make CROSS_COMPILE=${CROSS_TOOLS} -C $(CROSS_KBUILD_DIR) M=`pwd`

clean:
        KBUILD_NOPEDANTIC=1 make -C $(KBUILD_DIR) M=`pwd` clean

install:
        install -D -m 755 wl.ko $(MDEST_DIR)
sodo@DESKTOP-S799JDR:~/broadcom-wl$ export ARCH=arm
sodo@DESKTOP-S799JDR:~/broadcom-wl$ make cross
KBUILD_NOPEDANTIC=1 make CROSS_COMPILE=/home/sodo/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-11.2.0_musl_eabi/bin/arm-openwrt-linux- -C /home/sodo/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/linux-5.10.113/ M=`pwd`
make[1]: Entering directory '/home/sodo/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/linux-5.10.113'
CFG80211 API is prefered for this kernel version
Using Wireless Extension API
CFG80211 API is prefered for this kernel version
Using Wireless Extension API
  MODPOST /home/sodo/broadcom-wl/Module.symvers
ERROR: modpost: "global_var_42a8a4" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_421ef0" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_3d6" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_cf3" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_45fae5" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_8c6" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "__asm_rep_movsb_memcpy" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_424684" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_cf1" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_1061" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_421d90" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_42d7bc" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_421d10" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "__unnamed_28" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "__unnamed_69" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "__unnamed_26" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_45aaec" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "__unnamed_20" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "__unnamed_63" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "__unnamed_61" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "__unnamed_22" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "__unnamed_65" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "__unnamed_24" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "__unnamed_67" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_42b780" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_1c5" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_55b18f" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_42e4fc" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_14e" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_55b18d" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_42d884" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "__unnamed_39" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_42454c" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_3daf84" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_63a" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_45aab8" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_e9e" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_42dd4c" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "__unnamed_70" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_839" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_5b296c" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_55b191" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_42e438" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_58e" [/home/sodo/broadcom-wl/wl.ko] undefined!
[... 197 lines cut ...]
ERROR: modpost: "__unnamed_11" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "__unnamed_56" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_4242e4" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_421db5" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_ea0" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_42d3e0" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_42d380" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_42192c" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_3da8d8" [/home/sodo/broadcom-wl/wl.ko] undefined!
ERROR: modpost: "global_var_5afefb" [/home/sodo/broadcom-wl/wl.ko] undefined!
[... cut ...]
make[2]: *** [scripts/Makefile.modpost:124: /home/sodo/broadcom-wl/Module.symvers] Error 1
make[2]: *** Deleting file '/home/sodo/broadcom-wl/Module.symvers'
make[1]: *** [Makefile:1726: modules] Error 2
make[1]: Leaving directory '/home/sodo/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/linux-5.10.113'
make: *** [Makefile:161: cross] Error 2
```
this is the error i got when i just replaced wlc_hybrid with wl_sta.o

according to objdump wlc_sta and wlc_hybrid has many intersections in symbols(objdump -t)

antoineco, 2022-05-09:
I don't think I'll be able to help with this. `broadcom-wl` does **NOT** support ARM, neither does it support AP mode for modern hardware.

If you want to install a Linux-based system on a router, I recommend looking into [OpenWRT](https://openwrt.org/docs/techref/hardware/soc/soc.broadcom.bcm47xx) (which apparently you did).

Also, the `brcm80211` driver might be a better choice for you to run in AP mode.
The aforementioned page says:

> **brcm80211**
> The current version only supports station mode, but there are patches in current OpenWrt trunk which add support for AP and Ad Hoc mode.
>
> **broadcom-wl**
> [..] it contains some open source files, however binaries were pre-compiled for **x86 and x86_64 only**.
> So despite providing support for more hardware, it can't be integrated into OpenWrt because of **lack of support for MIPS/ARM** and important features like **AP mode**.

djdisodo, 2022-05-09:
let me just post information i gathered
there are three blobs here
https://github.com/Jackysi/advancedtomato/tree/advancedtomato-ac/release/src-rt-6.x/wl/linux
i guess sta stands for standard (client mode)
and ap stands for ap mode

and apsta for both

there's some source here
https://github.com/Jackysi/advancedtomato/tree/advancedtomato-ac/release/src-rt-6.x/linux/linux-2.6/drivers/net/wl
i don't understand much about it but
it's set to use prebuilt or build from source (wl/sys)
src-rt includes sys
src-rt-6.x doesn't

i tried to put three method trying to build this for arm

first was to put wl.ko itself(of older kernel) as wlc_hybrid.o
which arm version exists
since wl.ko is built from wlc_hybrid.o
i thought it may include wlc_hybrid.o

second was to put [lift-and-compiled version of wlc_hybrid.o](https://github.com/djdisodo/wl-recompile-experiment)

third was to put wl_sta.o from upper link

all three of them printed same error i pasted earlier

i'll do more research but i'm concerned that the error comes from something else than wlc_hybrid itself