antoineco/broadcom-wl #26: can not support AP mode.

fanlinbj126com, 2022-06-02:
lsapi
```
01:00.0 Network controller [0280]: Broadcom Inc. and subsidiaries BCM43224 802.11a/b/g/n [14e4:4353] (rev 01)
```

iw list
```
Wiphy phy0
        wiphy index: 0
        max # scan SSIDs: 1
        max scan IEs length: 0 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CMAC (00-0f-ac:6)
        Available Antennas: TX 0 RX 0
        Supported interface modes:
                 * IBSS
                 * managed
        Band 1:
                Bitrates (non-HT):
                        * 1.0 Mbps
                        * 2.0 Mbps (short preamble supported)
                        * 5.5 Mbps (short preamble supported)
                        * 11.0 Mbps (short preamble supported)
                        * 6.0 Mbps
                        * 9.0 Mbps
                        * 12.0 Mbps
                        * 18.0 Mbps
                        * 24.0 Mbps
                        * 36.0 Mbps
                        * 48.0 Mbps
                        * 54.0 Mbps
                Frequencies:
                        * 2412 MHz [1] (20.0 dBm)
                        * 2417 MHz [2] (20.0 dBm)
                        * 2422 MHz [3] (20.0 dBm)
                        * 2427 MHz [4] (20.0 dBm)
                        * 2432 MHz [5] (20.0 dBm)
                        * 2437 MHz [6] (20.0 dBm)
                        * 2442 MHz [7] (20.0 dBm)
                        * 2447 MHz [8] (20.0 dBm)
                        * 2452 MHz [9] (20.0 dBm)
                        * 2457 MHz [10] (20.0 dBm)
                        * 2462 MHz [11] (20.0 dBm)
                        * 2467 MHz [12] (20.0 dBm) (no IR)
                        * 2472 MHz [13] (20.0 dBm) (no IR)
                        * 2484 MHz [14] (20.0 dBm) (no IR)
        Band 2:
                Bitrates (non-HT):
                        * 6.0 Mbps
                        * 9.0 Mbps
                        * 12.0 Mbps
                        * 18.0 Mbps
                        * 24.0 Mbps
                        * 36.0 Mbps
                        * 48.0 Mbps
                        * 54.0 Mbps
                Frequencies:
                        * 5160 MHz [32] (30.0 dBm)
                        * 5170 MHz [34] (30.0 dBm)
                        * 5180 MHz [36] (30.0 dBm)
                        * 5190 MHz [38] (30.0 dBm)
                        * 5200 MHz [40] (30.0 dBm)
                        * 5210 MHz [42] (30.0 dBm)
                        * 5220 MHz [44] (30.0 dBm)
                        * 5230 MHz [46] (30.0 dBm)
                        * 5240 MHz [48] (30.0 dBm)
                        * 5250 MHz [50] (30.0 dBm)
                        * 5260 MHz [52] (30.0 dBm)
                        * 5270 MHz [54] (30.0 dBm)
                        * 5280 MHz [56] (30.0 dBm)
                        * 5290 MHz [58] (30.0 dBm)
                        * 5300 MHz [60] (30.0 dBm)
                        * 5310 MHz [62] (30.0 dBm)
                        * 5320 MHz [64] (30.0 dBm)
                        * 5330 MHz [66] (30.0 dBm)
                        * 5340 MHz [68] (30.0 dBm)
                        * 5350 MHz [70] (30.0 dBm)
                        * 5360 MHz [72] (30.0 dBm)
                        * 5370 MHz [74] (30.0 dBm)
                        * 5380 MHz [76] (30.0 dBm)
                        * 5390 MHz [78] (30.0 dBm)
                        * 5400 MHz [80] (30.0 dBm)
                        * 5410 MHz [82] (30.0 dBm)
                        * 5420 MHz [84] (30.0 dBm)
                        * 5430 MHz [86] (30.0 dBm)
                        * 5440 MHz [88] (30.0 dBm)
                        * 5450 MHz [90] (30.0 dBm)
                        * 5460 MHz [92] (30.0 dBm)
                        * 5470 MHz [94] (30.0 dBm)
                        * 5480 MHz [96] (30.0 dBm)
                        * 5490 MHz [98] (30.0 dBm)
                        * 5500 MHz [100] (30.0 dBm)
                        * 5510 MHz [102] (30.0 dBm)
                        * 5520 MHz [104] (30.0 dBm)
                        * 5530 MHz [106] (30.0 dBm)
                        * 5540 MHz [108] (30.0 dBm)
                        * 5550 MHz [110] (30.0 dBm)
                        * 5560 MHz [112] (30.0 dBm)
                        * 5570 MHz [114] (30.0 dBm)
                        * 5580 MHz [116] (30.0 dBm)
                        * 5590 MHz [118] (30.0 dBm)
                        * 5600 MHz [120] (30.0 dBm)
                        * 5610 MHz [122] (30.0 dBm)
                        * 5620 MHz [124] (30.0 dBm)
                        * 5630 MHz [126] (30.0 dBm)
                        * 5640 MHz [128] (30.0 dBm)
                        * 5650 MHz [130] (30.0 dBm)
                        * 5660 MHz [132] (30.0 dBm)
                        * 5670 MHz [134] (30.0 dBm)
                        * 5680 MHz [136] (30.0 dBm)
                        * 5690 MHz [138] (30.0 dBm)
                        * 5700 MHz [140] (30.0 dBm)
                        * 5710 MHz [142] (30.0 dBm)
                        * 5720 MHz [144] (30.0 dBm)
                        * 5725 MHz [145] (30.0 dBm)
                        * 5730 MHz [146] (30.0 dBm)
                        * 5735 MHz [147] (30.0 dBm)
                        * 5740 MHz [148] (30.0 dBm)
                        * 5745 MHz [149] (30.0 dBm)
                        * 5750 MHz [150] (30.0 dBm)
                        * 5755 MHz [151] (30.0 dBm)
                        * 5760 MHz [152] (30.0 dBm)
                        * 5765 MHz [153] (30.0 dBm)
                        * 5770 MHz [154] (30.0 dBm)
                        * 5775 MHz [155] (30.0 dBm)
                        * 5780 MHz [156] (30.0 dBm)
                        * 5785 MHz [157] (30.0 dBm)
                        * 5790 MHz [158] (30.0 dBm)
                        * 5795 MHz [159] (30.0 dBm)
                        * 5800 MHz [160] (30.0 dBm)
                        * 5805 MHz [161] (30.0 dBm)
                        * 5810 MHz [162] (30.0 dBm)
                        * 5815 MHz [163] (30.0 dBm)
                        * 5820 MHz [164] (30.0 dBm)
                        * 5825 MHz [165] (30.0 dBm)
                        * 5830 MHz [166] (30.0 dBm)
                        * 5840 MHz [168] (30.0 dBm)
                        * 5850 MHz [170] (30.0 dBm)
                        * 5860 MHz [172] (30.0 dBm)
                        * 5870 MHz [174] (30.0 dBm)
                        * 5880 MHz [176] (30.0 dBm)
                        * 5890 MHz [178] (30.0 dBm)
                        * 5900 MHz [180] (30.0 dBm)
                        * 5910 MHz [182] (30.0 dBm)
                        * 5920 MHz [184] (30.0 dBm)
                        * 5930 MHz [186] (30.0 dBm)
                        * 5940 MHz [188] (30.0 dBm)
                        * 5950 MHz [0] (30.0 dBm)
                        * 5960 MHz [2] (30.0 dBm)
                        * 5970 MHz [4] (30.0 dBm)
                        * 5980 MHz [6] (30.0 dBm)
                        * 5990 MHz [8] (30.0 dBm)
                        * 6000 MHz [10] (30.0 dBm)
                        * 6010 MHz [12] (30.0 dBm)
                        * 6020 MHz [14] (30.0 dBm)
                        * 6030 MHz [16] (30.0 dBm)
                        * 6040 MHz [18] (30.0 dBm)
                        * 6050 MHz [20] (30.0 dBm)
                        * 6060 MHz [22] (30.0 dBm)
                        * 6070 MHz [24] (30.0 dBm)
                        * 6080 MHz [26] (30.0 dBm)
                        * 6090 MHz [28] (30.0 dBm)
                        * 6100 MHz [30] (30.0 dBm)
                        * 6110 MHz [32] (30.0 dBm)
                        * 6120 MHz [34] (30.0 dBm)
                        * 6130 MHz [36] (30.0 dBm)
                        * 6140 MHz [38] (30.0 dBm)
        Supported commands:
                 * set_interface
                 * new_key
                 * join_ibss
                 * set_pmksa
                 * del_pmksa
                 * flush_pmksa
                 * connect
                 * disconnect
        software interface modes (can always be added):
        interface combinations are not supported
        Device supports scan flush.
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Supported extended features:
```

antoineco, 2022-06-02:
@fanlinbj126com that is unfortunately the case, and a limitation of this aging driver. `broadcom-wl` is capable of operating in AP mode, but only with old BCM47xx devices as far as I know.

Here is a good comparison of existing Broadcom drivers for Linux, with their tradeoffs: https://openwrt.org/docs/techref/hardware/soc/soc.broadcom.bcm47xx#broadcom-wl

antoineco, 2022-06-02:
@fanlinbj126com I just looked and the [brcm80211](https://wiki.debian.org/brcm80211) driver supports your device. Would that maybe allow you to enable the AP mode?

antoineco, 2022-07-07:
Closing due to inactivity.

closed 2022-07-07 by antoineco