antoineco/broadcom-wl #31: 6.0-rc1 needs patching

satmandu, 2022-08-15:
There's a version of a patch for arch available here: https://gist.github.com/joanbm/207210d74637870c01ef5a3c262a597d

But that doesn't apply cleanly to this.

satmandu, 2022-08-15:
Looks like a solution is here: https://github.com/supremegamers/broadcom-wl/commit/078a1afe3ebc5f70aaf7b17fc70d7fb57584ca26

[wl.diff](https://github.com/antoineco/broadcom-wl/files/9337623/wl.diff.txt)

```
--- src/wl/sys/wl_cfg80211_hybrid.c	2022-08-03 14:26:55.661367452 -0400
+++ /tmp/wl_cfg80211_hybrid.c	2022-08-15 09:59:34.322708286 -0400
@@ -2412,7 +2412,12 @@ wl_bss_roaming_done(struct wl_cfg80211_p
 	struct wl_cfg80211_connect_info *conn_info = wl_to_conn(wl);
 #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 12, 0)
 	struct cfg80211_roam_info roam_info = {
+// Rel. commit "cfg80211: Indicate MLO connection info in connect and roam callbacks" (Veerendranath Jakkam, Wed Jun 8)
+#if LINUX_VERSION_CODE < KERNEL_VERSION(6, 0, 0)
 		.bssid = wl->profile->bssid,
+#else
+		.links[0].bssid = wl->profile->bssid,
+#endif
 		.req_ie = conn_info->req_ie,
 		.req_ie_len = conn_info->req_ie_len,
 		.resp_ie = conn_info->resp_ie,
```

antoineco, 2022-08-15:
Thanks for the heads up! 🙌
I haven't had the occasion to try 6.0 yet. I'll make sure to patch and test before the release.

satmandu, 2022-08-15:
The aforementioned patch works for me:

```
broadcom-wl/6.30.223.271, 6.0.0-rc1, x86_64: installed
```

satmandu, 2022-08-15:
```
sudo dmesg | grep "Linux version \| wlan0"
[    0.000000] Linux version 6.0.0-rc1 (satadru@cheekon) (gcc (Ubuntu 11.3.0-5ubuntu1) 11.3.0, GNU ld (GNU Binutils for Ubuntu) 2.38.90.20220713) #1 SMP PREEMPT_DYNAMIC Sun Aug 14 20:00:14 EDT 2022
[   12.216401] wlan0: Broadcom BCM43a0 802.11 Hybrid Wireless Controller 6.30.223.271 (r587334)
[   12.225723] wl 0000:03:00.0 wlp3s0: renamed from wlan0
```

antoineco, 2022-08-15:
Closed by 933540c63f33e6ac2825d65c4b681ef3387d9146.

Thank you again for reporting and testing 👍

closed 2022-08-15 by antoineco