antoineco/broadcom-wl #32: WARNING: CPU: 4 PID: 409095 at net/wireless/sme.c:823 __cfg80211_connect_result+0x5ed/0x680 [cfg80211]

satmandu, 2022-08-18:
Not sure if anyone else is seeing this, but I've seen this with both the 5.19 and 6.0-rc1 kernels, and maybe earlier too...

```
[40089.502762] ------------[ cut here ]------------
[40089.502768] WARNING: CPU: 4 PID: 409095 at net/wireless/sme.c:823 __cfg80211_connect_result+0x5ed/0x680 [cfg80211]
[40089.502875] Modules linked in: rfcomm nls_utf8 cifs cifs_arc4 cifs_md4 fscache netfs qrtr cmac algif_hash algif_skcipher af_alg bnep nls_iso8859_1 intel_rapl_msr intel_rapl_common snd_hda_codec_cirrus x86_pkg_temp_thermal snd_hda_codec_generic intel_powerclamp ledtrig_audio snd_hda_codec_hdmi snd_hda_intel snd_intel_dspcfg snd_intel_sdw_acpi snd_hda_codec snd_hda_core snd_hwdep snd_pcm kvm_intel btusb snd_seq_midi snd_seq_midi_event btrtl btbcm btintel btmtk sch_fq_codel snd_rawmidi kvm joydev wl(POE) snd_seq mei_pxp mei_hdcp sbs rapl bluetooth cmdlinepart input_leds bcm5974 applesmc ecdh_generic snd_seq_device intel_cstate ecc sbshc spi_nor mei_me apple_mfi_fastcharge snd_timer cfg80211 snd mei soundcore acpi_als industrialio_triggered_buffer kfifo_buf apple_gmux industrialio apple_bl mac_hid facetimehd(OE) videobuf2_dma_sg videobuf2_memops videobuf2_v4l2 videobuf2_common videodev mc coretemp ipmi_devintf ipmi_msghandler msr parport_pc ppdev lp ramoops parport reed_solomon pstore_blk
[40089.502979]  pstore_zone mtd efi_pstore ip_tables x_tables autofs4 zfs(POE) zunicode(POE) zzstd(OE) zlua(OE) zavl(POE) icp(POE) zcommon(POE) znvpair(POE) spl(OE) z3fold lz4 lz4_compress hid_apple hid_generic usbhid hid i915 uas nouveau usb_storage mxm_wmi wmi drm_buddy i2c_algo_bit drm_ttm_helper ttm drm_display_helper cec rc_core drm_kms_helper syscopyarea crct10dif_pclmul sysfillrect crc32_pclmul ghash_clmulni_intel aesni_intel sysimgblt fb_sys_fops nvme spi_intel_platform crypto_simd spi_intel cryptd i2c_i801 drm thunderbolt nvme_core lpc_ich i2c_smbus xhci_pci xhci_pci_renesas video
[40089.503048] Unloaded tainted modules: mtdpstore():1 chromeos_pstore():1 mtdpstore():1 chromeos_pstore():1 fjes():1 mtdpstore():1 chromeos_pstore():1 asus_ec_sensors():1 fjes():1 asus_ec_sensors():1 mtdpstore():1 chromeos_pstore():1 fjes():1 fjes():1
[40089.503076] CPU: 4 PID: 409095 Comm: kworker/u16:0 Tainted: P           OE      6.0.0-rc1 #1
[40089.503082] Hardware name: Apple Inc. MacBookPro11,3/Mac-2BD1B31983FE1663, BIOS 432.60.3.0.0 10/27/2021
[40089.503085] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[40089.503144] RIP: 0010:__cfg80211_connect_result+0x5ed/0x680 [cfg80211]
[40089.503221] Code: 00 00 66 89 45 b8 e8 92 86 6a e6 48 8b 55 a8 e9 b0 fd ff ff 0f 0b 4c 89 ef e8 ff e0 ff ff 41 8b 14 24 85 d2 0f 85 0a ff ff ff <0f> 0b 4c 89 e6 4c 89 ef e8 a6 e6 ff ff e9 d2 fa ff ff 0f 0b 31 db
[40089.503225] RSP: 0018:ffffc2729ec6fd80 EFLAGS: 00010246
[40089.503229] RAX: 0000000000000000 RBX: ffff9e492d59b018 RCX: 0000000000000002
[40089.503232] RDX: 0000000000000000 RSI: 00000000fffffe00 RDI: ffff9e4890727000
[40089.503235] RBP: ffffc2729ec6fde0 R08: ffff9e48867960b8 R09: 0000000000000003
[40089.503238] R10: ffff9e492d59b3b0 R11: fffffffeff6855bb R12: ffff9e492d59b018
[40089.503240] R13: ffff9e4890727000 R14: ffff9e492d59b3b0 R15: 0000000000000000
[40089.503243] FS:  0000000000000000(0000) GS:ffff9e4bef300000(0000) knlGS:0000000000000000
[40089.503246] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[40089.503249] CR2: 00007ffacb1cf90b CR3: 000000042360f005 CR4: 00000000001706e0
[40089.503253] Call Trace:
[40089.503255]  <TASK>
[40089.503260]  cfg80211_process_wdev_events+0x115/0x190 [cfg80211]
[40089.503323]  ? cfg80211_process_wdev_events+0x115/0x190 [cfg80211]
[40089.503384]  cfg80211_process_rdev_events+0x2b/0x40 [cfg80211]
[40089.503445]  cfg80211_event_work+0x2a/0x40 [cfg80211]
[40089.503501]  process_one_work+0x219/0x3f0
[40089.503508]  worker_thread+0x50/0x3e0
[40089.503513]  ? process_one_work+0x3f0/0x3f0
[40089.503517]  kthread+0xeb/0x110
[40089.503524]  ? kthread_complete_and_exit+0x20/0x20
[40089.503531]  ret_from_fork+0x1f/0x30
[40089.503540]  </TASK>
[40089.503542] ---[ end trace 0000000000000000 ]---
```

antoineco, 2022-08-20:
@satmandu what are the messages preceding this trace?

satmandu, 2022-08-21:
Now I'm getting a different issue...

```
5] task:kworker/u16:35  state:R  running task     stack:    0 pid:688250 ppid:     2 flags:0x00004008
[91362.607029] Workqueue: events_unbound async_run_entry_fn
[91362.607034] Call Trace:
[91362.607035]  <TASK>
[91362.607039]  ? wlc_bmac_read_shm+0xe/0x10 [wl]
[91362.607091]  wlc_init+0x314/0x1053 [wl]
[91362.607126]  ? wlc_ampdu_tx_reset+0x8b/0x109 [wl]
[91362.607163]  wl_init+0x74/0x80 [wl]
[91362.607208]  wlc_up+0x20f/0x3dc [wl]
[91362.607242]  wl_up+0x38/0x80 [wl]
[91362.607288]  wl_resume+0x52/0x100 [wl]
[91362.607334]  pci_pm_resume+0x68/0xf0
[91362.607337]  ? pci_legacy_resume+0x80/0x80
[91362.607339]  dpm_run_callback+0x6c/0x170
[91362.607341]  device_resume+0xad/0x200
[91362.607343]  async_resume+0x1e/0x40
[91362.607344]  async_run_entry_fn+0x2d/0x110
[91362.607346]  process_one_work+0x219/0x3f0
[91362.607348]  worker_thread+0x50/0x3e0
[91362.607350]  ? process_one_work+0x3f0/0x3f0
[91362.607351]  kthread+0xeb/0x110
[91362.607354]  ? kthread_complete_and_exit+0x20/0x20
[91362.607356]  ret_from_fork+0x1f/0x30
```
[dmesg.txt](https://github.com/antoineco/broadcom-wl/files/9389989/dmesg.txt)

satmandu, 2022-08-21:
Wifi link has weird dropouts at times...

closed 2024-02-04 by antoineco