antoineco/broadcom-wl #38: Failing to create a hotspot using TP Link Archer T6E AC1300 (BCM4360)

tukykarmakar, 2023-12-02:
I'm using a Fedora-based distro ([Nobara](https://nobaraproject.org/)) with KDE, dual-booting with Windows 10 from a physically separate drive. I have the **TP Link Archer T6E AC1300** wifi card and **broadcom-wl** driver version _6.30.223.271-21.fc38_ from the _RPM Fusion Non-free_ repository. My card is working fine and connects with my router, hotspots from my phones, but is unable to create a hotspot from the desktop.

Every time I click on the Hotspot button from the Network menu in the system tray, nothing happens and I get the message "**Authorisation supplicant timed out**". Below the is the notification popup that I see--
[image]
**Blacksite** is the name of _my PC_ and **wlp4s0** is the _wifi card name_ in the system.

Below is the output of the command `lspci -vnn -d 14e4:`--
```
04:00.0 Network controller [0280]: Broadcom Inc. and subsidiaries BCM4360 802.11ac Wireless Network Adapter [14e4:43a0] (rev 03)
        Subsystem: Broadcom Inc. and subsidiaries Device [14e4:0619]
        Flags: bus master, fast devsel, latency 0, IRQ 31
        Memory at fc400000 (64-bit, non-prefetchable) [size=32K]
        Memory at fc200000 (64-bit, non-prefetchable) [size=2M]
        Capabilities: <access denied>
        Kernel driver in use: wl
        Kernel modules: bcma, wl
```
I know that my **card supports the AP mode** because I can create hotspots in Windows 10, and also, it shows in the `iw list` output--
```
Supported interface modes:
                 * IBSS
                 * managed
                 * AP
```
I ran `journatctl -f` and saved the logs in the attached file. [journalctl.txt](https://github.com/antoineco/broadcom-wl/files/13536502/journalctl.txt)

I tried using [linux-wifi-hotspot](https://github.com/lakinduakash/linux-wifi-hotspot) by following [this](https://github.com/lakinduakash/linux-wifi-hotspot#installation-1) instruction, but I keep getting the following error--
```
pkexec --user root create_ap wlp4s0 enp5s0 'Blacksite' '12345678' --mkconfig /etc/create_ap.conf --no-virt  
Config options written to '/etc/create_ap.conf'

Error: Failed to run hostapd, maybe a program is interfering.
If an error like 'n80211: Could not configure driver mode' was thrown
try running the following before starting create_ap:
    nmcli r wifi off
    rfkill unblock wlan

Command not found or exited with error status
```
PC Specifications:-
- CPU: AMD Ryzen 5 3600
- GPU: AMD Radeon RX 6600 8GB GDDR6
- Motherboard: ASrock B450M Pro 4
- RAM: 2x8 GB 3600 MHz running at 3200 MHz
- Storage: 1 TB NVMe m.2, 1 TB SATA 3 6 GB/s m.2, 1 TB SATA 3 6 GB/s 2.5'
- Power: Corsair RM650 80+ Gold

I have tried restarting the wpa_supplicant and the NetworkManager services multiple times, reinstalled the driver, tried using pre-installed drivers but card wasn't detected. Now I have no idea why the WPA Supplicant is failing.

I'm new to Linux, but I can find my way around the system, so I'll appreciate all the details provided when preforming tasks for troubleshooting. If it's not possible to create access point at all, then I'll appreciate the explanations too.

Thanks!

antoineco, 2023-12-02:
Unfortunately this aging driver does not support AP mode: https://github.com/antoineco/broadcom-wl/issues/26#issuecomment-1144690708

tukykarmakar, 2023-12-03:
Well that sucks! My fried gifted me this card seeing that it was one of the most popular cards on Amazon. At least I can connect to the internet and if needed, I'll boot up Windows I guess.

Is there any guides or resources that'll help a new user in Linux to buy a supported wifi card?

Am I correct in assuming that TP Link gets the chips from Broadcom and creates the firmware based on the firmware provided by Broadcom? I'll be opening a support case with TP Link and let them know about this. I know that it won't fix the issues, but it should put the info in the back of their head. I urge everyone having the same issue in Linux and reading this, to open a support case too. I'll update here with their response.

antoineco, 2023-12-03:
The card is perfectly fine, it's the support for it in Linux which is suboptimal because Broadcom seems to only release proprietary drivers for Windows and macOS, and the OSS options do not support all Broadcom chips.

I ran into the same issue as you did (minus the support for AP mode, which I don't use). This was in fact what prompted me to start maintaining this distro-agnostic patch repo.

closed 2023-12-05 by antoineco