antoineco/broadcom-wl #40: BCM4360 - Ubuntu doesn't seems to use the driver

moi15moi, 2023-12-05:
I have a ``Rosewill RNX-AC1300PCE`` wifi-card which use BCM4360.
I installed this package by following the manual instruction in the readme (so the make installation).
I rebooted my computer, but ubuntu doesn't recognize my wifi-card like you can see in this screenshot:
[image]

I don't know if it is revelant, but I also got some warning when I did a ``make``. I don't know if it change anything:
```
moi15moi@moi15moi-System-Product-Name:~/Téléchargements/broadcom-wl-patch-linux4.7$ make
KBUILD_NOPEDANTIC=1 make -C /lib/modules/`uname -r`/build M=`pwd`
make[1] : on entre dans le répertoire « /usr/src/linux-headers-6.2.0-26-generic »
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-11 (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0
  You are using:           gcc-11 (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
CFG80211 API is prefered for this kernel version
Using CFG80211 API
  CC [M]  /home/moi15moi/Téléchargements/broadcom-wl-patch-linux4.7/src/shared/linux_osl.o
  CC [M]  /home/moi15moi/Téléchargements/broadcom-wl-patch-linux4.7/src/wl/sys/wl_linux.o
  CC [M]  /home/moi15moi/Téléchargements/broadcom-wl-patch-linux4.7/src/wl/sys/wl_iw.o
  CC [M]  /home/moi15moi/Téléchargements/broadcom-wl-patch-linux4.7/src/wl/sys/wl_cfg80211_hybrid.o
  LD [M]  /home/moi15moi/Téléchargements/broadcom-wl-patch-linux4.7/wl.o
CFG80211 API is prefered for this kernel version
Using CFG80211 API
  MODPOST /home/moi15moi/Téléchargements/broadcom-wl-patch-linux4.7/Module.symvers
  CC [M]  /home/moi15moi/Téléchargements/broadcom-wl-patch-linux4.7/wl.mod.o
  LD [M]  /home/moi15moi/Téléchargements/broadcom-wl-patch-linux4.7/wl.ko
  BTF [M] /home/moi15moi/Téléchargements/broadcom-wl-patch-linux4.7/wl.ko
Skipping BTF generation for /home/moi15moi/Téléchargements/broadcom-wl-patch-linux4.7/wl.ko due to unavailability of vmlinux
make[1] : on quitte le répertoire « /usr/src/linux-headers-6.2.0-26-generic »
```

Here is the details of my Network controller:
```
moi15moi@moi15moi-System-Product-Name:~$ sudo lspci -vnn -d 14e4:
[sudo] Mot de passe de moi15moi : 
04:00.0 Network controller [0280]: Broadcom Inc. and subsidiaries BCM4360 802.11ac Wireless Network Adapter [14e4:43a0] (rev 03)
	Subsystem: Broadcom Inc. and subsidiaries BCM4360 802.11ac Wireless Network Adapter [14e4:0619]
	Flags: bus master, fast devsel, latency 0, IRQ 17
	Memory at f7400000 (64-bit, non-prefetchable) [size=32K]
	Memory at f7200000 (64-bit, non-prefetchable) [size=2M]
	Capabilities: [48] Power Management version 3
	Capabilities: [58] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [68] Vendor Specific Information: Len=44 <?>
	Capabilities: [ac] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [13c] Device Serial Number 68-1c-00-ff-ff-00-00-00
	Capabilities: [150] Power Budgeting <?>
	Capabilities: [160] Virtual Channel
	Capabilities: [1b0] Latency Tolerance Reporting
	Capabilities: [220] Physical Resizable BAR
	Kernel driver in use: bcma-pci-bridge
	Kernel modules: bcma, wl
```

antoineco, 2023-12-05:
The README file contains all the instructions. Please read them.

antoineco, 2023-12-05:
The `wl` module was loaded just fine.

If you disable your WiFi interface and re-enable it (e.g. by restarting the network service) do you see any relevant error reported by the `dmesg` command?

moi15moi, 2023-12-05:
The output it too long, so I put it in a text file [Output.txt](https://github.com/antoineco/broadcom-wl/files/13561469/Output.txt)

But, I don't see any apparent error

antoineco, 2023-12-05:
The `wl` module "taints" the kernel according to the logs, but that shouldn't be a blocker.

Possibly the `bcma` module wasn't blacklisted properly and your device is trying to use it:
```
[    0.883240] bcma-pci-bridge 0000:04:00.0: bus0: Bus registered
```

In your original message you wrote:
```
	Kernel driver in use: bcma-pci-bridge
	Kernel modules: bcma, wl
```

moi15moi, 2023-12-05:
> Possibly the `bcma` module wasn't blacklisted properly and your device is trying to use it:

Here is my ``/etc/modprobe.d/blacklist.conf`` file. Like you can see at the end of it, I wrote ``blacklist bcma``, so I guess it should be good, no?

```
moi15moi@moi15moi-System-Product-Name:~$ cat /etc/modprobe.d/blacklist.conf
# This file lists those modules which we don't want to be loaded by
# alias expansion, usually so some other driver will be loaded for the
# device instead.

# evbug is a debug tool that should be loaded explicitly
blacklist evbug

# these drivers are very simple, the HID drivers are usually preferred
blacklist usbmouse
blacklist usbkbd

# replaced by e100
blacklist eepro100

# replaced by tulip
blacklist de4x5

# causes no end of confusion by creating unexpected network interfaces
blacklist eth1394

# snd_intel8x0m can interfere with snd_intel8x0, doesn't seem to support much
# hardware on its own (Ubuntu bug #2011, #6810)
blacklist snd_intel8x0m

# Conflicts with dvb driver (which is better for handling this device)
blacklist snd_aw2

# replaced by p54pci
blacklist prism54

# replaced by b43 and ssb.
blacklist bcm43xx

# most apps now use garmin usb driver directly (Ubuntu: #114565)
blacklist garmin_gps

# replaced by asus-laptop (Ubuntu: #184721)
blacklist asus_acpi

# low-quality, just noise when being used for sound playback, causes
# hangs at desktop session start (Ubuntu: #246969)
blacklist snd_pcsp

# ugly and loud noise, getting on everyone's nerves; this should be done by a
# nice pulseaudio bing (Ubuntu: #77010)
blacklist pcspkr

# EDAC driver for amd76x clashes with the agp driver preventing the aperture
# from being initialised (Ubuntu: #297750). Blacklist so that the driver
# continues to build and is installable for the few cases where its
# really needed.
blacklist amd76x_edac

# wireless drivers (conflict with Broadcom hybrid wireless driver 'wl')
blacklist ssb
blacklist bcma
blacklist b43
blacklist brcmsmac
```

antoineco, 2023-12-05:
Let's check a few things:

1. Blacklisted entries are only considered after a reboot. Did you reboot?
1. To unload modules which were already loaded, you have to call `rmmod` (with `sudo`). But since you said you already did it, I'm not sure what the problem could be. Are you certain that `sudo rmmod bcma` returned without error?
1. Maybe check your `/etc/modules` file and ensure that there is `bcma` entry in there?

moi15moi, 2023-12-05:
> * Blacklisted entries are only considered after a reboot. Did you reboot?

Yes, I did reboot.

> * To unload modules which were already loaded, you have to call `rmmod` (with `sudo`). But since you said you already did it, I'm not sure what the problem could be. Are you certain that `sudo rmmod bcma` returned without error?

It doesn't print anything when I run ``sudo rmmod bcma``. Is that a good sign?

> 3\. Maybe check your `/etc/modules` file and ensure that there is no `bcma` entry in there? (or any of the other blacklisted modules)

I don't have a folder named ``/etc/modules``.
I verified all the file in ``/etc/modprobe.d/`` and none of them (except of course ``blacklist.conf``) blacklist ``bcma``.

antoineco, 2023-12-05:
What happens if you `rmmod wl` and `modprobe wl` again? As before, please check `dmesg` for clues and verify with `sudo lspci -vnn -d 14e4` what driver is in use.

moi15moi, 2023-12-05:
Here is the output of these command (I runned all of them has root): [Output.txt](https://github.com/antoineco/broadcom-wl/files/13572126/Output.txt)

It seems to be pretty much the same has the previous output I share of those commands

moi15moi, 2023-12-10:
I found a solution.

I runned the command ``sudo update-initramfs -u`` and rebooted. Now, it doesn't ignore the blacklist for bcma.

I can now connect to my local router.

It may be a good idea to mention it in the Readme. What do you think?

moi15moi, 2023-12-10:
Strange, I just rebooted and now, ubuntu doesn't detect my wifi-card again.

antoineco, 2023-12-10:
Interesting, I had never heard of the blacklist file being in the initramfs before. If that's a standard thing on Ubuntu I should definitely mention it indeed.

I have personally not experienced instabilities of that sort, but `dmesg` will show you any error relating to the wl module.

antoineco, 2023-12-10:
I see that the Debian package is doing exactly what you did upon installation: https://salsa.debian.org/broadcom-sta-team/broadcom-sta/-/blob/debian/debian/broadcom-sta-common.postinst (learned something!)

They are also blacklisting a few extra modules but I'm not sure that it matters: https://salsa.debian.org/broadcom-sta-team/broadcom-sta/-/blob/debian/debian/broadcom-sta-common.modprobe

moi15moi, 2023-12-10:
> I have personally not experienced instabilities of that sort, but `dmesg` will show you any error relating to the wl module.

Here is the output: [output.txt](https://github.com/antoineco/broadcom-wl/files/13627812/output.txt)
I don't see any error message.

> Side question: since Ubuntu also has this package, why not simply install it? This would save you the effort of re-building the module on every kernel update, or setting up dkms manually.

Because I'm too far from my router to use an ethernet cable.

> They are also blacklisting a few extra modules, maybe you should try that as well: https://salsa.debian.org/broadcom-sta-team/broadcom-sta/-/blob/debian/debian/broadcom-sta-common.modprobe

I tried, but it doesn't correct the problem.

antoineco, 2023-12-10:
From the kernel logs alone, everything is looking fine to me:
```
[    3.967072] wl 0000:04:00.0: enabling device (0000 -> 0002)
[    4.038135] wlan0: Broadcom BCM43a0 802.11 Hybrid Wireless Controller 6.30.223.271 (r587334)
[    4.199668] wl 0000:04:00.0 wlp4s0: renamed from wlan0
```

> When I boot, I need to forget the wifi and restart the network because it won't connect to my router via `sudo systemctl restart systemd-networkd`.

That is more of an OS configuration issue than a kernel module issue. I don't have enough information to help here, but it could be that the network is managed through the `NetworkManager` service (one level on top of systemd).

> Also, it seems that the wifi only work for a few minute and then, it deconnect from the wifi (but ubuntu still can see the wifi)

If it works and then disconnects without emitting any kernel message, it means that the issue isn't caused by a failure of the kernel module. In case I was right about `NetworkManager` being in charge of the network on Ubuntu, I would suggest checking the journal logs of that service around the time the network disconnected.

moi15moi, 2023-12-10:
Here is the output of ``sudo journalctl -fu NetworkManager`` when I just booted and wait for ~5 minutes
[output.txt](https://github.com/antoineco/broadcom-wl/files/13628103/output.txt)

moi15moi, 2023-12-10:
I found the problem. It was my fault.

In brief, I just needed ``sudo update-initramfs -u`` to correct my problem.

closed 2023-12-10 by moi15moi